dokuhn/openDTU-BreakoutBoard #2: PCB's available for sale?

bjoerns1983, 2022-11-17:
Hi there,
i love your idea, because i was just about designing the same but as far not as professional as yours because i have never done that before.
Just have two questions:
1. Dow you have some boards for sales, i would need 3 units.
2. Would is be possible possible to change the position of the cap so that also higher ones could be mounted
3. In some forums i have read that beside the elctrolyte cap also a ceramic one should be place infront of the NRF.

bjoerns1983, 2022-11-18:
I would to take the two board you have left over, or is there a gerber file etc. which only needs to be uploaded to jlpcb etc. to get them manufactured.
Sorry if the question might be dumb i have never done that before.

Really love the simplicity of the adapter.

dokuhn, 2022-11-18:
I can send you two boards and the right capacitors for some small bucks. If necessary I will order more boards soon. That depends on the requirements in my network with friends and colleagues.

bjoerns1983, 2022-11-19:
That would be awesome, simply tell me what you get incl. shipping in Germany.
Are there private messages on Github so would could clearify the details?

dokuhn, 2022-11-20:
You can send me a mail at [dominik.kuhn90@googlemail.com](mailto:dominik.kuhn90@googlemail.com), then we can coordinate details about shipping etc.

dirkheiden, 2022-11-26:
It’s not very difficult to order these PCB from i.e. Aisler. The capacitors can be ordered on eBay for example.
[image]

bjoerns1983, 2022-12-06:
Okay after waiting for ages for the Pinheaders it done:
[image]

bjoerns1983, 2022-12-19:
If anybody is interested: I have ordered some boards myself and have some left over.
So if anyone wants to try it just drop me a mail to: 3ddruck@sengotta.net

MoritzLerch, 2022-12-24:
Old vs. New

Thanks to @bjoerns1983 for the PCB 😄 

[image]

bjoerns1983, 2022-12-30:
Okay all gone here.

evilmumi, 2023-01-11:
Hi, are some PCB still available to buy?
Or can some1 tell me where / how i can order this best from .de
I only need one

Archibald-Harry-Tuttle, 2023-01-11:
I have some left.
Added mounting holes on all 4 corners for 3mm screws.
just write me pn pls.

evilmumi, 2023-01-11:
> I have some left. Added mounting holes on all 4 corners for 3mm screws. just write me pn pls.

Hi, a pn on github is not possible or?

Archibald-Harry-Tuttle, 2023-01-11:
> > I have some left. Added mounting holes on all 4 corners for 3mm screws. just write me pn pls.
> 
> Hi, a pn on github is not possible or?

You find me in the discord server of OpenDTU repository.

guha7, 2023-01-12:
Does anybody one PCB left?
Thanks for your offer!

Archibald-Harry-Tuttle, 2023-01-12:
I have some left.
like in the last post before,
you find me on discord.

webwerb, 2023-01-18:
Sorry, ich bin zu blöd den Ansprechpartner zu finden ....

Ich bräuchte insgesamt 3 Platinen für folgende Konfiguration
- Espressif ESP32 WLAN Dev Kit Board Development Bluetooth Wifi v1 WROOM32 Node MCU
- NRF24L01+ PLUS - PA LNA SMA Antenne Long Range Funk Modul Raspberry Arduino
- 0,96 Zoll OLED Display I2C SSD1306 

Ev. auch 3 Gehäuse für Nutzung mit Display, da ich noch keinen Drucker habe.

Wäre dankbar für entsprechende Tips für einen Anfänger hier auf der Platform.

Danke!

dokuhn, 2023-01-18:
Hi, you are not too stupid to find the contact person. 

Thank you for your request. I am the contact person for this repo, but as you may have seen, this is a community driven project. That means members of the community support each other with for example boards or even questions about variants etc.. In the "home" README file is described for which board this breakout board is intended. Is your board compatible with it ? There are also other variants with less pins which can be used by a small hack of the board. For this you can have a look at the corresponding issue. Otherwise @Archibald-Harry-Tuttle  as you can see above probably still board left. You can reach him under the following [Discord link](https://discord.gg/WzhxEY62mB). I currently have no more boards left, new ones are on the way and will take more than 10 days to arrive.

Dominik Kuhn

Pixel-Chris, 2023-01-26:
@dokuhn 
Hi, sind die Boards reinzufällig schon angekommen?

webwerb, 2023-01-26:
Moin,

 

gerade eingetroffen.

 

Super schnell – Danke!

 

Ich Dussel hab natürlich vorher nicht geschaut ob 30 oder 38 Pin und ich hab natürlich die 30 Pin Version vom ESP32 ….

 

Das wird dann so nicht passen?

 

Also 30 Pin Borads kaufen (wo?) oder 38 Pin ESP 32 kaufen. Dann könnte ich gleich welche mit externer Antenne fürs WLAN nehmen.

 

Grüße

 

Uwe

dokuhn, 2023-02-01:
Nachdem in Fernost auch das neue Jahr vorbei ist, hab ich eine Mail bekommen, dass die nächsten Platinen unterwegs sind zu mir. Sprich die 20 Platinen des normalen Breakout-Board werden demnächst dann eintreffen. Die SMD Variante ist in den letzten Zügen und wird in den nächsten Wochen ebenfalls als Prototypen mal aufgebaut.

dirkheiden, 2023-02-01:
Auf die SMD Version freue ich mich sehr. Hast Du dafür schon eine BOM?

wiedimot, 2023-02-11:
> 

@dokuhn wie kann man die Boards denn bei dir bestellen?
Ich hätte auch interesse an 3 Stück. In dem Fall die "normale" Version des Breakout Boards.

dokuhn, 2023-02-12:
@wiedimot aktuell hab ich von der normalen Variante noch gut 15 Stück "vorrätig". Du kannst du mir einfach eine persönliche Mail schreiben. Darin kannst du mir dann auch die Adresse etc. mitteilen wo das Ganze hin soll. In der Mail können wir dann klären ob du mir etwas überweist oder den PayPal Button nutzen möchtest. 

Auch wenn meine Platinen bei mir mal temporär aus sind, ist es so, dass die Community sich da behilflich sehr behilflich ist. Ich werde entsprechenden Link die Tage mal erweitern in der README damit die ganzen Channel auch gefunden werden können.

andreas-bulling, 2023-02-17:
Hallo zusammen - ich hätte auch Interesse an einem Bord. Die normale Version würde mir reichen - sind noch welche da?

dokuhn, 2023-02-17:
Wie in der Message zuvor beschrieben bitte ein persönliche  E-Mail an mich, dann können wir Details wie Adresse etc. dort auf dem kurzen Weg klären. 

### Kurzes Update:  
Aktuell sind noch rund **6-7 Boards** übrig.  Ich würde nochmal welche Bestellen. Ansonsten bitte nicht wundern, ich besorge jetzt keine hunderte der Boards in der Phase, in der ich Beschaffungen tätige für die SMD/SMT Variante. Aus diesem Grund gibt es meiner Seitz lediglich Losgrößen zwischen 20 - 50 Platinen.

huckes, 2023-02-20:
Hallo, 
hatte auch Interesse an drei Boards per PM vor ein paar Tagen bekundet, leider bisher keine Antwort. Mail ist evtl. untergegangen oder ich bin zu ungeduldig?

dokuhn, 2023-02-20:
### **Update:**

Die letzte Bestellung ist jetzt mehr oder weniger kommplett raus. Sprich das letzte BreakoutBoard liegt schon im Hausgang für den Weg zur Post. Ich danke allen für die Unterstützung des Projekts an dieser Stelle. Die nächsten Boards sind aktuell wohl in der Produktion und werden vorausichtlich Ende nächster Woche eintreffen. 

The last order is now more or less completely shipped. The last BreakoutBoard is already in the corridor on its way to the post office. I would like to thank everyone for their support of the project. The next boards are currently in production and will probably arrive at the end of next week.

dokuhn, 2023-03-16:
### **Update:**

Die neue Variante ist mal prototypisch aufgebaut und wird getestet. Auch wenn es inzwischen schon merklich länger hell ist abends, könnte das Bild besser sein um 21 Uhr. ;-) 

[image]

dokuhn, 2023-03-21:
Ich freue mich das die neue Variante auf Interesse stößt. In Summe sind jetzt 4 Boards aufgebaut und werden durch unterschiedliche Kollegen der Community und im Umkreis des Vereins auf Herz und Nieren getestet. Danach (in den nächsten 1 - 2 Wochen) wird es ein Batch geben welches wir fertig bestückt bei Aisler und Co. Beschaffen. [image]