dokuhn/openDTU-BreakoutBoard #3: Specs of C1?

dersch81, 2022-11-17:
Hi, what are the specs of C1?

dokuhn, 2022-11-18:
Hi, i used a capacitor with 22uF and 50V voltage resistance, I had many of them at home. But a capacitor with 10uF and 6.3V is a good choice.

stefan123t, 2022-11-28:
We usually recommend 47-100uF on the Discord but that is just word of mouth. I guess anything between 10 and 100uF should be sound.
@dokuhn thanks for the board!
Are there plans to leave a bit more space for the Elko ?
currently it does not fit easily between the MCU and NRF boards. 
Thanks time and again for your contribution !

If you change the layout it may be good to have the six Test-Pins under the board as removable solder bridges so someone could use a different Pinout between MCU and NRF.

closed 2023-03-09 by dokuhn