dokuhn/openDTU-BreakoutBoard #5: 3D Print Case

bjoerns1983, 2022-12-13:
Just as info i have designed a 3D Printable Case for the Breakout board. If the next testprint is okay, i will send you a PR with the STL and Fusion 360 Files.

[image]
[image]

Bonze255, 2022-12-16:
holes for wall mounting on the underside would still be good .. :)

dokuhn, 2022-12-16:
Thank you for your feedback. The case looks very cool. I will adjust the file structure so that KiCAD files and CAD etc. are separated from each other in directories. This makes it easier with the PR and the linking in the README file.

bjoerns1983, 2022-12-18:
#7 
Okay PR is created. I have no added any hole for mounting because i dont want the screws to short anything.
Prints quite fast. If any modification is neede the fusion 360 files are included.

bjoerns1983, 2023-01-10:
Yeah no problem, simply tell me what you need.

CodeFinder2, 2023-02-03:
A version with an OLED display cutout would also be cool :)

bjoerns1983, 2023-02-04:
I dont have such a board. If you have on, grab the Fusion360 Files, modify them, testprint them and send a PR.

huckes, 2023-02-20:
Hello, I droped a remix of Bjoerns design with an opening for 1.3" OLED display to Thingiverse: https://www.thingiverse.com/thing:5866231

CodeFinder2, 2023-02-20:
Looks like it is an 0,96" display. Do you have a link to the display you used/showed in the photo?

Thank you very much for sharing! 😊

bjoerns1983, 2023-02-20:
> Hello, I droped a remix of Bjoerns design with an opening for 1.3" OLED display to Thingiverse: https://www.thingiverse.com/thing:5866231 [image]

Cool, maybe you could drop a PR so it could be added to the Repo?

huckes, 2023-02-20:
Hi,

@CodeFinder2**:  it is a 1.3" display, rather fast delviered from China: https://de.aliexpress.com/item/4001145494936.html?spm=a2g0o.order_list.order_list_main.19.7e2a5c5fQHz6fJ&gatewayAdapt=glo2deu

@bjoerns1983 : Currently it's just a Q&D remix. I try to refine it a bit (e.g. covering the lower part of the glass where nothng is displayed). When more or less final, I'll drop a PR. In the meanwhile the lid is available under Thingiverse.

I also want to test it completely set up with the hardware, as I'm not certain if the pins of the display come in conflict with the ESP-module.

bjoerns1983, 2023-02-20:
Ah okay thats why it is tagged WIP.

dokuhn, 2023-02-20:
very cool job, i am very happy to see cool extensions to the project. Feel free to post a PR when it's tested so we can add it to the repo. This way a lot of people will get something from it.