dokuhn/openDTU-BreakoutBoard #8: Mention the boards that fit to the layout

erniberni, 2022-12-23:
It would be wise to mention the boards that fits to the layout
Users should compare the pinout before connecting
This one is fine -> https://www.az-delivery.de/products/esp32-developmentboard
This one doesn't fit -> https://www.mischianti.org/2021/02/17/doit-esp32-dev-kit-v1-high-resolution-pinout-and-specs/
Take a special look at the connection of VCC and GND

dokuhn, 2022-12-27:
Thank you for the hint. I will add the corresponding entry to the README file at the next time. Additions or maintenance are welcome.

andy5macht, 2022-12-31:
I used ESP32 DEV KIT v1 and had to modify/bridge something. here is what i did:
first use these pins for ESP32 DEV KIT v1:
    -DHOYMILES_PIN_MISO=18
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=5
    -DHOYMILES_PIN_IRQ=4
    -DHOYMILES_PIN_CE=2
    -DHOYMILES_PIN_CS=17

Cut D19 from ESP32 DEV KIT v1 (otherwise D19 has short circuit to GND)
Scratch GND next to GND Pin of ESP32 DEV KIT v1 to connect GND of this BreakoutBoard to ESP32 DEV KIT v1.
Solder 3V3 Pin of ESP32 DEV KIT v1 to Elko (positive side)
If all done this should work an look like this:
[image]
[image]

stepl, 2023-01-18:
Does any of you could help me to understand what's difference between these 2 boards and if they fit?
- [ESP-32 Dev Kit C V4](https://www.az-delivery.de/products/esp-32-dev-kit-c-v4?variant=30871551869024)
- [ESP32 NodeMCU](https://www.az-delivery.de/products/esp32-developmentboard?variant=36542176914)

From the pin layout to me it's look like both fit the layout but would still be good to get a confirmation and to understand what differences are between the NodeMCU and the V4 version?

jelli123, 2023-01-18:
The second one is the "ESP32 Dev Kit C V2". Beside some improvements it should work like the first one.
[Schematic V4](https://cdn.shopify.com/s/files/1/1509/1638/files/ESP-32_NodeMCU_Developmentboard_Schematic_korr.pdf?v=1666000508)
[Schematic V2](https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf)