dokuhn/openDTU-BreakoutBoard #11: Wemos D1 compatible Version

bjoerns1983, 2023-02-07:
Hi there,
i know many people are waiting for the SMD Version of the adapter, which looks like a really cool project.
But to be honest i dont think that could be assambled at home any more, and if i look at other projects like ebusd, when chip shortage hits you there is a problem.

On the other hand we have a really simple adapter board which could be assambled at home using components you can buy for under 10€ at big commercial resellers. Thats really DIY and there was no shortage of these components in the past years.

A problem that occured with the adapter board was that there are boards that looks similiar, if you dont look really close, but they are not the same.

Long story short. The formfactor of the Lolin/Wemos D1 Mini was constant about the last few years. There are many clones of this board but they all share the same formfactor and pinout. There is also a D1 Mini32 which features the ESP32 Chipset.
https://www.wemos.cc/en/latest/d1/d1_mini.html
https://www.az-delivery.de/products/esp32-d1-mini 

So what i like to ask: Would it be possible to adapt the standard Breakout Board to that formfactor and maybe add a Pinheader for Displays that are mentioned by the OpenDTU Project?

bjoerns1983, 2023-02-20:
@dokuhn any Idea if that would be a doable solution?
I would do it myself but i have ne experience regarding Kicad.

dokuhn, 2023-02-20:
Sorry for the late feedback. I was a little busy with the ordering of the parts for the SMT variant etc. I like your idea and can totally imagine that we will expand this as soon as the testing of the SMT variant is completed.

thomic, 2023-03-07:
just my 2 cents, I think D1Mini isn't a bad form factor, but a "breakout board" has so many downsides, like broken boards (PINs) or connection issues after some time - so in the end - it makes a lot of sense to have a DIY solution based on the SMT/SMD version
Just get a board, possibly pre-assembled already, and start flashing it.
In my mind, it should be as easy as flashing a Tasmota device or WLED onto an existing device. Not having to a lot to solder or break before.
Sure, the breakout board will still be alive and I guess will see some iterations, but as you need to have at least some soldering skills and like 1 hour to assemble everything, it won't be that much straight forward as a SMT variant.