dokuhn/openDTU-BreakoutBoard #12: Add a OLED Version

dersch81, 2023-02-22:
Hi, since OpenDTU is now supporting displays I suggest to add an additional Display version of the Case and PCB to have the connector breakouts as well as mounting points with a cut out of the case lid. 

I would support a cheap an common one like this: https://www.amazon.de/dp/B01L9GC470/ref=pe_27091401_487024491_TE_item

dersch81, 2023-02-24:
Ok, i have noticed the case lid with space for the OLED: https://www.thingiverse.com/thing:5866231

So why not adding it here? And just modify the PCB to have a Pinout to connect the OLED?

huckes, 2023-02-24:
Hi, please find the updated version here: https://www.thingiverse.com/thing:5866226.

I also changed the case, as the ESP-boards I had in stock are shorter and more slim as the boards for which the case originally was designed. I'll state a PR for the PCB soon.

SteffMUC, 2023-02-28:
Just fyi, spent some time to add the display connection directly on the PCB, case is currently in preparation, however not yet perfectly fitting, nevertheless you can see the intermediate status and concept here:
https://github.com/SteffMUC/openDTU_wDisplay2

thomic, 2023-03-07:
This is something the SMT (or maybe even the breakout variant) of the board can feature in a future edition. At the moment the focus is a bit on getting the SMT version done and production ready, just to get boards to the people. Still it should be possible to add a display to the board later if you got some time, soldering skills and interest. But for now -as time for pet projects is limited- I guess OLED by default will not be there in the first round.